tesseract-robotics/trajopt #428: Questions Regarding Convex Solver in Trajopt

lihelin666, 2024-10-06:
I would like to express my gratitude for the amazing work done by the Tesseract Robotics team. While using Trajopt, I've encountered a few questions that I hope you could help clarify (please forgive me if any of these questions seem naive):

**1. What convex solver is used in the tesseract_planning\tesseract_examples?**

I came across a report  [https://rosindustrial.org/news/2018/7/5/optimization-motion-planning-with-tesseract-and-trajopt-for-industrial-applications](https://rosindustrial.org/news/2018/7/5/optimization-motion-planning-with-tesseract-and-trajopt-for-industrial-applications) The article mentions: “The remaining application to discuss is a complex Semi-Constrained Cartesian path with 437 poses each with 5 Degrees Of Freedom (DOF) fixed and the tool z-axis free to rotate. The application is performing a deburr operation on a complex puzzle piece shown in Figure 4. Also, the problem includes a 7 DOF robot and a 2 DOF positioner for the spindle, making it a non-fixed base kinematic chain. This requires the use of Tesseract’s joint kinematic model developed for this particular use case. The TrajOpt motion planner problem contains roughly 3,000 constraints and was able to solve in 4.4 seconds.”

I have rewritten the puzzle_piece_example using Tesseract-Python on Windows, and it takes around 10 minutes to run. I am curious about the computer configuration and the specific convex solver used to achieve the 4.4 seconds mentioned in the report.

**2. What convex solver is used in Tesseract-Python? Is it locked, or can it be customized?**

Has there been a quantitative comparison of different convex solvers' performance on various examples? Which one is recommended?

BPMPD (interior point method, free for non-commercial use only)
Gurobi (simplex and interior point/parallel barrier, license required)
OSQP (ADMM, BSD2 license)
qpOASES (active set, LGPL 2.1 license)

**3. How does ifopt fit into Trajopt? I feel it operates parallel to the convex solvers mentioned above. Is it important for the future of the overall project?**

I have noticed similar statements like “if (ifopt_){...}else{...}” in almost all Trajopt-related demos, suggesting that there might be a dedicated API developed specifically for ifopt. I'm curious about its role within the entire project.

**4. There is a perspective that “TrajOpt is a local optimizer for trajectories. If it is likely that a local planner will not find a direct path, an OMPL planner can be used to find a clear path, and then TrajOpt can optimize the path.”** (see https://tesseract-robotics.github.io/tesseract_python/_source/modules/tesseract_motion_planners_trajopt/tesseract_motion_planners_trajopt.html). In practice, for multi-axis planning problems (like puzzle_piece_example and car_seat_example), using OMPL to find an initial solution may also fail, while directly using TrajOpt takes a long time. Are there any effective strategies to address this issue?

**5. Does the Descartes planner have limitations when planning for 6-7 axis problems?** If the provided tool_path is relative to the base coordinate system, it can plan successfully. However, if the tool_path is relative to J6 or J7 (as in the puzzle_piece_example), it fails. Is there a restriction in the DescartesPlanner regarding this?

**6. Is there any consideration for releasing a precompiled version of the entire Tesseract Robotics project for Windows?** I noticed it is an important part of the Scan N Plan project. Additionally, how can one apply for access to join https://ros-win.visualstudio.com/ros-win/_build?

lihelin666, 2024-10-12:
Dear @Levi-Armstrong , I would greatly appreciate it if you could offer some assistance. Thank you so much!

Levi-Armstrong, 2024-10-15:
1. What convex solver is used in the tesseract_planning\tesseract_examples?

    Most if not all of them will use QSQP. The default value can be found by reviewing the TrajOptDefaultSolverProfile. In the case of the puzzle piece it explicitly set it to OSQP.

2. What convex solver is used in Tesseract-Python? Is it locked, or can it be customized?

    @johnwason Do you know the answer to this question?

3. How does ifopt fit into Trajopt? I feel it operates parallel to the convex solvers mentioned above. Is it important for the future of the overall project?

   The original TrajOpt was written specifically for joint trajectories and nearly impossible to introduce additional optimization variables. As a result it has been rewritten using the IFOPT framework being called TrajOptIfopt as a replacement for TrajOpt to address limitations of the original implementation. The solve time is not as good as the original in some cases which is why the original implementation has not been removed. I have tracked down the issue and have a started MR to address the root cause but I have not had time to finish the work. Once this is finished it should be complete and will be significantly faster than the original implementation. 

4. There is a perspective that “TrajOpt is a local optimizer for trajectories. If it is likely that a local planner will not find a direct path, an OMPL planner can be used to find a clear path, and then TrajOpt can optimize the path.”

OMPL should always find a solution given enough time, but that may not be optimal. I usually create a task composer pipeline which runs several different pipelines in parallel and picks the first one that finishes successfully. A lot of this really depends on the requirements of the system to decide the best approach. 

5. Does the Descartes planner have limitations when planning for 6-7 axis problems?

   The main limitation, is as you start to increase the axes over 6, the problem increases significantly in size requiring a lot of RAM. 

6. Is there any consideration for releasing a precompiled version of the entire Tesseract Robotics project for Windows?

I am interested in this and we put a lot of work on making this possible, but need someone that is familar with windows and NuGet to finalize and work out any remaining issues. Every package should be configured to leverage CPACK to generate binaries for both linux and windows. This can be enabled using -DTESSERACT_PACKAGE=ON.

johnwason, 2024-10-15:
> What convex solver is used in Tesseract-Python? Is it locked, or can it be customized?

All the solvers are built on Linux, and OSQP and QPOASES are available on Windows. I actually don't know what one is loaded by default. @Levi-Armstrong is there a yaml configuration parameter somewhere for which solver is used?

> Is there any consideration for releasing a precompiled version of the entire Tesseract Robotics project for Windows?

For Python this has been available for a long time. You can use pip to install the `tesseract-robotics` package. See the readme: https://github.com/tesseract-robotics/tesseract_python?tab=readme-ov-file#installation

I have been working on a binary release of Tesseract on Windows using conda on and off for a long time with variable levels of success. This package includes the developer files for C++ so you can build against it. If you are adventurous you can try it out: https://anaconda.org/Tesseract-Robotics/tesseract-robotics-superpack

rjoomen, 2024-10-16:
Regarding 1):
- [Default solver for Trajopt.](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_motion_planners/trajopt/include/tesseract_motion_planners/trajopt/profile/trajopt_default_solver_profile.h#L49)
- [Available solvers for Trajopt.](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_sco/include/trajopt_sco/solver_interface.hpp#L229-L236)
- [Trajopt-Ifopt currently only supports OSQP.](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_motion_planners/trajopt_ifopt/include/tesseract_motion_planners/trajopt_ifopt/profile/trajopt_ifopt_default_solver_profile.h#L59-L61)