tesseract-robotics/trajopt #430: Configuration limitation with safety_margin and safety_margin_buffer

Levi-Armstrong, 2024-10-26:
I recently ran into an failure issue using UR where the default collision config for trajopt sets the safety_margin=0.025 and the safety_margin_buffer=0.05 but there is one link pair on the UR which are always within 0.035. As a result this results with a lot of collision equation being added to the problem causing issues. This can be solved by reducing the safety margin_buffer to but that also can cause solver issues if it too small. Now trajopt supports setting individual link pair safety_margin data but not the safety_margin_buffer which needs to be added to provide necessary configuration. For now you can reduce the things such that the safety_margin + the safety_margin_buffer is less than the 0.035 but may not work for all cases.

rjoomen, 2024-10-26:
Yes, I did wonder why it was not possible to set the buffer individually. Are you considering adding this?

Levi-Armstrong, 2024-10-26:
I think it should be added, but not sure when I would be able to get to it.

rjoomen, 2024-10-26:
I'll take a quick look on Monday how much work this would be, but if it's not trivial, I don't think I have time to do it anytime soon, either.

rjoomen, 2024-10-28:
I suppose the buffer should be added [here](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_common/include/trajopt_common/utils.hpp#L72), right? (And all dependent code should be adapted to that, of course.)

Levi-Armstrong, 2024-10-30:
> I suppose the buffer should be added [here](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_common/include/trajopt_common/utils.hpp#L72), right? And remove [this](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt/include/trajopt/problem_description.hpp#L632), as it will be part of [info](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt/include/trajopt/problem_description.hpp#L639). (And all dependent code should be adapted to that, of course.)

I believe your assessment is correct.

Levi-Armstrong, 2024-10-30:
Maybe a new struct instead of the array?