tesseract-robotics/trajopt #444: This might be a bug

liuminL, 2025-01-18:
This will cause LVSDiscrete to not perform any interpolation

Levi-Armstrong, 2025-01-18:
The user defines the longest valid segment fraction or length which is used for interpolation. If the distance between the two states is already smaller than this threshold, interpolation is not needed.

liuminL, 2025-01-19:
> The user defines the longest valid segment fraction or length which is used for interpolation. If the distance between the two states is already smaller than this threshold, interpolation is not needed.

But this should be a function of the LVSDiscreteCollisionEvaluator class. If the user specifies the configuration type as tesseract_collision::CollisionEvaluatorType::LVS_DISCRETE, interpolation will never be performed, which has nothing to do with the value of
longest_valid_segment_length. But if the user specifies the type as LVS_CONTINUOUS, it should call the CalcCollisionsHelper function of the LVSContinuousCollisionEvaluator class, so is there any problem with this?

Levi-Armstrong, 2025-01-19:
@liuminL Can you test #445 and see if resolves your issue?

closed 2025-01-19 by liuminL