tesseract-robotics/trajopt #452: Linking error: Relocations in generic ELF (EM: 62) for bpmpd_linux64.a(bpmdll.o) in tesseract_ws/trajopt/trajopt_sco

liuminL, 2025-02-24:
# Problem Description

During the compilation of a project based on the tesseract_ws workspace, a linking error occurred. Specifically, when linking the bpmdll.o object file from the tesseract_ws/trajopt/trajopt_sco/3rdpartylib/bpmpd_linux64.a library, the linker (/usr/bin/ld) threw the error message: Relocations in generic ELF (EM: 62).
# System and Environment Information

    - Operating System: Ubuntu 22.04 LTS
    - Architecture: AArch64
    - Compiler: gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 
    - CMake Version: 3.22.1
# Expected Solution
I hope to obtain or recompile the bpmpd library suitable for the AArch64 architecture to resolve the current architecture incompatibility issue, enabling the project to compile and run normally. 

**Thank you in advance for your support and guidance!**

Levi-Armstrong, 2025-02-24:
I do not think it is possible to leverage this on AArch64

liuminL, 2025-02-25:
Thank you very much, I disabled this solver

closed 2025-02-25 by liuminL